adafruit/Adafruit_CircuitPython_Display_Text #8: Label initalization very slow with bdf font

rdagger, 2019-04-13:
Initializing a label takes about 10 seconds with a bitmap font.  Here's my code:
```
font = bitmap_font.load_font('terminalpants.bdf')
text_area = label.Label(font, text='Hello world')  # VERY SLOW ~ 10 seconds
text_area.x = 10
text_area.y = 10
display.show(text_area)
```
Here's the bdf font:  https://github.com/pixelmasochist/terminalpants

tannewt, 2019-04-15:
It's parsing the file the first time it needs a glyph (to save memory.) Use `font.load_glyphs` to preload the glyphs you need all in one pass of the file. Source is here: https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/blob/master/adafruit_bitmap_font/glyph_cache.py#L53

closed 2019-04-15 by tannewt

zouden, 2019-12-29:
This doesn't work like I expect it to.
```

font_large = bitmap_font.load_font('/fonts/Arial-BoldMT-90.bdf')
font_large.load_glyphs('0123456789.-') 
lbl = label.Label(font=font_large, text="0.00", x=20, y=200, max_glyphs=5)

```

My problem: updating the `lbl.text` with new values takes a signficant amount of time the first time, but then the glyphs are cached and it runs fast. I want to pre-cache the glyphs. `load_glyphs()` doesn't work like I expect it to. What am I doing wrong?

tannewt, 2020-01-02:
I'd expect that to work. To debug, I'd add a print of characters here: https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/blob/master/adafruit_bitmap_font/bdf.py#L100

zouden, 2020-01-02:
I should have mentioned in my previous comment that `font_large.load_glyphs('0123456789.-') ` takes a noticeable amount of time, which suggests it is doing something. It just doesn't speed up the label text updating.

tannewt, 2020-01-03:
What new values cause the slow update? Can you time a subsequent update of a known label with time.monotonic()? It'd be slow if the text accidentally included a new character.

zouden, 2020-01-05:
Here's my test code for timing. It omits the label construction.

```
font_large = bitmap_font.load_font('/fonts/Arial-BoldMT-90.bdf')
info_target_speed_left.font = font_large
info_target_speed_left.text = "0.00" 

starttime = time.monotonic()
font_large.load_glyphs('0123456789.-')
print(time.monotonic() - starttime)

starttime = time.monotonic()
info_target_speed_left.text = "1.23" 
print(time.monotonic() - starttime)

starttime = time.monotonic()
info_target_speed_left.text = "4.56"
print(time.monotonic() - starttime)

starttime = time.monotonic()
info_target_speed_left.text = "1.23"
print(time.monotonic() - starttime)
```

Output:
```
1.55908 //cache
1.125 //changing to 1.23
1.36401 //changing to 4.56
0.00390625 //changing to 1.23
```
Now if I comment out the cache line:
```
0.0
1.36914
1.36475
0.00341797
```

So yeah it seems that load_glyphs takes time but doesn't actually cache. Yet label caching _does_ work because the last line is faster than the others.

tannewt, 2020-01-07:
Good find @zouden. `load_glyphs` didn't handle strings correctly. I've add support for them here: https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/pull/15 This will screw up alternate encodings but I think that is ok. Everyone should use UTF-8 now.