adafruit/Adafruit_CircuitPython_Display_Text #11: Group full when assigning longer text

caternuson, 2019-05-23:
Maybe I'm incorrectly assuming the mutability of a Label? You can reassign the text, but if it is longer than the original text, then you exceed a Group limit.

```python
Adafruit CircuitPython 4.0.0 on 2019-05-20; Adafruit PyPortal with samd51j20
>>> import terminalio
>>> from adafruit_display_text import label
>>> label = label.Label(terminalio.FONT, text="123456789")
>>> label.text = "1"
>>> label.text = "1234"
>>> label.text = "1234567"
>>> label.text = "123456789"
>>> label.text = "1234567890"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_display_text/label.py", line 174, in text
  File "adafruit_display_text/label.py", line 119, in _update_text
RuntimeError: Group full
>>>
```

iraytrace, 2019-06-03:
Given the nature of the Label, I would suggest this should either truncate the input text in the property setter method, or otherwise have a more graceful recovery.

anecdata, 2019-06-03:
I believe this behavior is currently as-designed. Label needs the maximum number of characters, or pixel width using the bounding box using the current font. Enlarging the allocation, auto-truncation option, even a scrolling text option would be great. This gets tricky with multi-line text though.

makermelissa, 2019-07-04:
If you pass in `max_glyphs=30` or whatever number is your largest size, then it will handle this correctly. If not specified, it assumes the length of the default text is the max_glyphs. I believe this issue is more about having it degrade with a better message since "Group full" is a little misleading.

dhalbert, 2019-07-31:
I just encountered this too. I'd suggest either it just truncates or we catch the "Group full" exception and throw with a more meaningful exception message like "Text too long" or "Text exceeds max_glyphs".

caternuson, 2019-07-31:
Agree with above two comments - this issue is just a suggestion for better fail behavior. And once implemented, update the displayio guide.

Would truncating be the more user friendly option?

kevinjwalters, 2020-03-01:
I just got confused with this too. I agree with @dhalbert and @jepler in Discord, a re-throw with a clearer exception would be a good thing.

kevinjwalters, 2020-06-26:
`max_glyphs` is also problematic as this is not documented in enough detail for the user to know that this needs to be `maximum(all possible text lengths) + 1` - note the `+ 1`.

The `raise RuntimeError("Please provide a max size, or initial text")` in the code doesn't really help here either as it mentions a `max size`.

caternuson, 2020-07-15:
Closing this for now. Looks like it was nominally fixed with #48.

closed 2020-07-15 by caternuson