adafruit/Adafruit_CircuitPython_Display_Text #13: MemoryError: memory allocation failed

hulkco, 2019-07-08:
I have this issue, when I use the adafruit_bitmap_font library in conjunction with Adafruit_CircuitPython_Display_Text
This the program
```
# Write your code here :-)
import os
import board
import displayio
from adafruit_display_text.label import Label
from adafruit_bitmap_font import bitmap_font
from adafruit_st7735r import ST7735R

spi = board.SPI()
tft_cs = board.D5
tft_dc = board.D6
rst = board.D7

displayio.release_displays()
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs, reset=rst)
display = ST7735R(display_bus, width=160, height=83, colstart=24, rowstart=1, rotation=90)
#display = ST7735R(display_bus, width=160, height=80, colstart=24, rotation=90)

# the current working directory (where this file is)
cwd = ("/"+__file__).rsplit('/', 1)[0]
fonts = [file for file in os.listdir(cwd+"/fonts/")
         if (file.endswith(".bdf") and not file.startswith("._"))]
for i, filename in enumerate(fonts):
    fonts[i] = cwd+"/fonts/"+filename
print(fonts)

##########################################################################
THE_FONT = fonts[0]
DISPLAY_STRING = "A multi-line-\nexample of\n  font bounding!"
WRAP_CHARS = 20

##########################################################################
# Make the display context
splash = displayio.Group()
display.show(splash)

# Make a background color fill
color_bitmap = displayio.Bitmap(160, 80, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0xFFFFFF
bg_sprite = displayio.TileGrid(color_bitmap,
                               pixel_shader=color_palette,
                               x=0, y=0)
splash.append(bg_sprite)

# Load the font
font = bitmap_font.load_font(THE_FONT)
font.load_glyphs(DISPLAY_STRING.encode('utf-8'))

print(DISPLAY_STRING)

text = Label(font, text=DISPLAY_STRING)
text.x = 20
text.y = 40
text.color = 0x0

# Make a background color fill
dims = text.bounding_box
print(dims)
textbg_bitmap = displayio.Bitmap(dims[2], dims[3], 1)
textbg_palette = displayio.Palette(1)
textbg_palette[0] = 0xFF0000
textbg_sprite = displayio.TileGrid(textbg_bitmap,
                                   pixel_shader=textbg_palette,
                                   x=text.x+dims[0], y=text.y+dims[1])
splash.append(textbg_sprite)
splash.append(text)
board.DISPLAY.refresh_soon()
board.DISPLAY.wait_for_frame()

while True:
    pass

```

and this is the error.

```
>>> import text3
['/fonts/Helvetica-Bold-16.bdf', '/fonts/Alef-Bold-18.bdf', '/fonts/Arial-12.bdf', '/fonts/Comic-Bold-18.bdf', '/fonts/Noto-18.bdf', '/fonts/Verdana-Bold-18.bdf']
A multi-line-
example of
  font bounding!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "text3.py", line 52, in <module>
  File "adafruit_display_text/label.py", line 79, in __init__
  File "adafruit_display_text/label.py", line 114, in _update_text
MemoryError: memory allocation failed, allocating 92 bytes
>>> 

```

ladyada, 2019-07-08:
M0/samd21 boards don't have a ton of RAM, you probably need to upgrade to an M4 (nRF/D51)

hulkco, 2019-07-09:
Thanks @ladyada !!!

closed 2019-07-09 by ladyada