adafruit/Adafruit_CircuitPython_Display_Text #15: display_text_simpletest.py is blank, not functional.

jtrip, 2019-07-21:
Hello, while looking into this library the first time I noticed that the display_text_simpletest.py is blank. The README.md file illustrates a simple test that I think would be great, however I would also like to suggest making one small improvement.

Currently the usage example is:  
```
import board
import terminalio
from adafruit_display_text import label

text = "Hello world"
text_area = label.Label(terminalio.FONT, text=text)
text_area.x = 10
text_area.y = 10
board.DISPLAY.show(text_area)
```

Which does seem to work in my testing, however after updating the display with the text_area object the code immediately ends which leads to circuitpython displaying the message that the code is done running. My only concern is that beginners might not have the intuition to think that it is working fine but some additional control needs to be used in order for it to be visible for more than a moment, and then the beginner might be chasing down a ghost thinking it isn't working right.  

I think a beginner friendly improvement would be to add a delay so that they should be able to see the output but then also see that when the code is done executing it will be replaced by the circuit python "Code done running" message.   
Like so:  
```
import board
import terminalio
from adafruit_display_text import label
from time import sleep

text = "Hello world"
text_area = label.Label(terminalio.FONT, text=text)
text_area.x = 10
text_area.y = 10
board.DISPLAY.show(text_area)
sleep(3)
```
Another simple option might be to put the board.DISPLAY.show call in a loop (maybe infinite like `while True:`) but maybe that's not setting a great example for beginners or is otherwise adding irrelevant logic to the example.

I would be happy to make these changes and submit a pull request. I don't think I have submitted a pull request to any for your repos yet so I thought I would introduce the idea here. Perhaps there is some reason that example is currently missing or if you have any thoughts about the change please let me know.

ladyada, 2019-07-21:
oops looks like we missed that example, please feel free to submit a nice simple one. what youv've got looks great. instead of a sleep, please do a 'while True: pass' type loop so it stays on screen (or you can bounce the text around :D

jtrip, 2019-07-22:
With that merged in I think we are good here, thanks!

closed 2019-07-22 by jtrip