adafruit/Adafruit_CircuitPython_Display_Text #27: Docs link in readme doesn't work

jepler, 2019-12-11:
It takes me to a [page](https://circuitpython.readthedocs.io/projects/display_text/en/latest/) that says 
```
        \          SORRY            /
         \                         /
          \    This page does     /
           ]   not exist yet.    [    ,'|
           ]                     [   /  |
           ]___               ___[ ,'   |
           ]  ]\             /[  [ |:   |
           ]  ] \           / [  [ |:   |
           ]  ]  ]         [  [  [ |:   |
           ]  ]  ]__     __[  [  [ |:   |
           ]  ]  ] ]\ _ /[ [  [  [ |:   |
           ]  ]  ] ] (#) [ [  [  [ :===='
           ]  ]  ]_].nHn.[_[  [  [
           ]  ]  ]  HHHHH. [  [  [
           ]  ] /   `HH("N  \ [  [
           ]__]/     HHH  "  \[__[
           ]         NNN         [
           ]         N/"         [
           ]         N H         [
          /          N            \
         /           q,            \
        /                           \
```

siddacious, 2019-12-22:
Assuming there isn't an issue building the docs, is it's probably just a matter of adding it to readthedocs.org. I can do this next week

tannewt, 2020-01-02:
I've fixed this by deleting the `display-text` subproject and recreating it as `display_text`. Thanks for the issue!

closed 2020-01-02 by tannewt