adafruit/Adafruit_CircuitPython_Display_Text #55: anchor_point and anchored_position do not work well with scale

kevinjwalters, 2020-06-27:
I updated my libraries (from `adafruit-circuitpython-bundle-5.x-mpy-20200625`) and tried this,

```
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>> import terminalio, displayio, board
>>> from adafruit_display_text import label
>>> text_area = label.Label(terminalio.FONT, text="Scale!", scale=5, color=0xc0c0c0)
>>> board.DISPLAY.show(text_area)
>>> text_area.anchor_point
(0, 0)
>>> text_area.anchored_position
(0, -7)
>>> text_area.anchored_position = (0, 0)  ### expectation is text will be on screen at top left
```

The position of that text is not entirely on screen, the top is chopped off. This is presumably due to issues with a `scale` which isn't the default value of `1`. Discussed this a little bit recently with @FoamyGuy in Discord.

FoamyGuy, 2020-06-27:
I think the root of this issue is that the `scale=5` parameter doesn't really belong to the Label, and Label isn't taking it into account for anything or really doing anything with it, except passing it through to the super class Group `__init__()` here: https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/blob/1dcecb9979dced10f3d4bf1fa7c4dc8e7c699f71/adafruit_display_text/label.py#L87 
using some arcane pythonic magic involving `**kwargs`

I think in order for `anchor_point` and `anchored_position` to play nicely with scaled text we need to capture the scale factor and keep it inside of Label and use it to when we calculate the new location.

I will try to take an attempt at making these changes.

makermelissa, 2020-07-02:
Fixed by #57.

closed 2020-07-02 by makermelissa