adafruit/Adafruit_CircuitPython_Display_Text #60: Changing the label text causes a change in the text position

kmatch98, 2020-07-02:
Using the example file `display_text_background_color_padding.py` the text "none" is changed to say "text".  In the process, the "text" label changes position and walks down the screen.

I think that either the getter or setter for anchored_position is not working properly.

kevinjwalters, 2020-07-03:
For libs from `adafruit-circuitpython-bundle-5.x-mpy-20200703` (`VERSIONS` file tells me that's `2.7.6`), I spotted something similar whilst investigating the _second issue_ in [Adafruit Forums: Clue_Plotter Problem](https://forums.adafruit.com/viewtopic.php?f=65&t=165178), the one reported by Robertl. I first tried commented out the **Button Guide** in the main code which appeared to act as viable workaround for a `MemoryError` but then the text labels on the axis ticks wandered off the screen gradually as they changed.

kmatch98, 2020-07-03:
Thanks for the comment about the other related error. I think the recent fixes will repair the “wandering off” of labels when the text is updated, but I don’t understand the Memory Error that was indicated in the forum post. Any ideas about the root cause of that?

If there is another problem I’d like to be sure we consider it with any next updates (and try not to create any new ones!)

kevinjwalters, 2020-07-03:
I put in #61 for the `MemoryError`.

kmatch98, 2020-07-09:
Resolved in https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/pull/59 
and https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/pull/62

closed 2020-07-09 by kmatch98