adafruit/Adafruit_CircuitPython_Display_Text #65: List index out of range

makermelissa, 2020-07-09:
It turns out the issue with skipping the glyphs with zero-length tilegrids results in a List index out of range error:
```
Traceback (most recent call last):
  File "electioncal.py", line 58, in <module>
    gfx.elections_cycle()
  File "/home/pi/electioncal_graphics.py", line 86, in elections_cycle
    self.election_name_text_line2.text, self.election_name_text.text = self.paragrapher(self.electioncal["dates"][i]["name"], 40)
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_display_text/label.py", line 362, in text
    self._update_text(str(new_text))
  File "/home/pi/.local/lib/python3.7/site-packages/adafruit_display_text/label.py", line 292, in _update_text
    self[i].position = (position_x, position_y)
  File "/home/pi/.local/lib/python3.7/site-packages/displayio/group.py", line 142, in __getitem__
    return self._layers[index]
IndexError: list index out of range
```

Perhaps a better solution would be to check the width and height of the glyph and if zero, just substitute a 1. That essentially how I was doing it as a work-around, though it's possible that may cause text alignment issues. Another option may be just putting a `None` in there for face as an empty placeholder and checking for that when the text is rendered.

kmatch98, 2020-07-09:
I tried a quick change to put `None` as a face, but got a different error.  I think we need to rethink the strategy around this one.

When updating text in a tileGrid, the code checks to see if any characters are changed and only updates the ones that are different.  Now that we deleted any zero-width tileGrids, the indexing between the text array and the tileGrid Group is off when it goes to update the tileGrid.  I think the best solution will be to just regenerate all the tileGrids, even if they are the same character.  It may have some performance penalty, but I think it won't be noticeable for what most folks are doing.

I'll propose a change and then I'll request your  feedback.

makermelissa, 2020-07-09:
Fixed by #66.

closed 2020-07-09 by makermelissa