adafruit/Adafruit_CircuitPython_Display_Text #68: Label disappears after text parameter is changed more than once.

AdinAck, 2020-07-09:
When I create a label, and then update it's text parameter later, the text will completely disappear upon the second change.

code:
`label = Label(terminalio.FONT, x=10, y=32, text="Test1", max_glyphs=5, color=palette[15],)
group.append(label)

time.sleep(1)
label.text = "Test2"
time.sleep(1)
label.text = "Test3"
time.sleep(1)
label.text = "Test4"
time.sleep(1)`

After "Text2" It will disappear, although, sometimes I will see "Test3" For a fraction of a second.

Perhaps I am using this wrong, in which case any help would be much appreciated!

dherrada, 2020-07-10:
@AdinAck Can you show me your whole code.py? You actually need a loop to keep displayio running so the code doesn't finish.

dherrada, 2020-07-10:
That being said, you should be seeing test3 and test4

makermelissa, 2020-07-10:
@AdinAck please try version 2.7.7. If that works, this may be related to issue #67.

AdinAck, 2020-07-10:
The update fixed the issue!
Thanks!

closed 2020-07-10 by AdinAck