adafruit/Adafruit_CircuitPython_Display_Text #72: Don't allocate a background bitmap when background_color is None

tannewt, 2020-07-22:
I'm getting this even when background_color is None.

```
  File "code.py", line 50, in <module>
  File "code.py", line 47, in <module>
  File "adafruit_pyportal.py", line 1028, in fetch
  File "adafruit_pyportal.py", line 601, in set_text
  File "adafruit_display_text/label.py", line 123, in __init__
  File "adafruit_display_text/label.py", line 295, in _update_text
  File "adafruit_display_text/label.py", line 205, in _update_background_color
  File "adafruit_display_text/label.py", line 162, in _create_background_box
MemoryError: memory allocation failed, allocating 3780 bytes
```

FoamyGuy, 2020-07-22:
I think this might be resolved with #71. I think `_create_background_box` is not getting called if `background_color` is `None` with those changes.

FoamyGuy, 2020-07-22:
Nope I was wrong. That is still getting called when background_color is None if custom font is used.

Thank you. I will get this fixed.

PR #74 (2020-07-24): dont create a background bitmap if background_color is None
I think this will resolve #72
closed 2020-07-24 by tannewt