adafruit/Adafruit_CircuitPython_Display_Text #76: Reduce memory usage for text creation: bitmap_label

kmatch98, 2020-07-28:
In order to reduce the amount of memory for displaying text, @foamyguy and I are developing a new "label" class that will use a bitmap instead of a group of TileGrids to store the text.  Here is a proposed strategy for this new "bitmap_label" Class:

### `bitmap_label` Class definition
* Input parameters:
    * Make `text` a required parameter (Note: This will flag if someone creates a bitmap_label with no text, which doesn’t make any sense.)
    * For direct compatibility with label, keep the `max_glyphs` parameter, but ignore it.
* Do we want to keep self._memorySaver?  (Note: If so, update this to a class variable)
    * True - do not save the text value
    * False - save the text value
    * If we keep this option, then we need to create a getter for text, returns `None` if self._memorySaver == True.
* Since the bitmap_label has limited edit capability after creation, minimize the instance variables that are stored
* Add getters/setters for:
     - anchor_point 
     - anchor_position
     - color
     - background_color
* Add scale capability

### General steps for `_init_` of a bitmap_label instance:
1. Error-checking: If text is “”, raise an exception.
2. Calculate the text box size, including any newlines and tabs
3. Calculate the background box size, including padding
4. Determine the bitmap size required by comparing the text box and background box max x,y dimensions.
5. Determine start location of text in the bitmap
6. Initialize the palette, TileGrid (self), Bitmap
7. Draw the background fill, as required
8. Place the text in the bitmap (Note: We can combine background and text writing into the same for-loops)
9. Set the TileGrid x,y placement based on anchor_point and anchor_position.

tannewt, 2020-07-28:
Is the core problem with the current implementation the overhead of the TileGrid class? How does this change as the text gets larger? The TG model saves glyph bitmap duplication at the expense of the TG fixed overhead and the cost of storing it in a group.

Note, I wouldn't include the current background bitmap in the calculations because it's easy to optimize on it's own.

kmatch98, 2020-07-28:
Short answer to your question:
### Conclusion: `label.py` uses >100 bytes per character.

---------------
#### Here are the details
Per your questions, I did a trial between `label.py` and my `TextMap` library (https://github.com/kmatch98/CircuitPython_textMap). 

For this trial, I used a string (659 characters long).  I printed one additional character during each loop and then measured the memory after each 10 characters are added.  I compared two fonts (Helvetica Bold 16 and Vera Sans Roman 24).  

For your reference, my TextMap example creates a bitmap the size of the screen (320x240 in this case) and copies each glyph's bits into the bitmap.

I measured:
- Memory available before starting the loop of printing the characters (to estimate "overhead"), using `gc.collect()` then `gc.mem_free()`
- Memory reduction after each character is added (bytes per added character)

#### Sample text output from `label.py` usage with Helvetica Bold 16:
```
After display.show(myGroup), just before loop start  Memory free: 85232
charCount: 10, Running Memory free: 84208, mem usage in Loop: 1024, per character mem: 93.0909
charCount: 20, Running Memory free: 82944, mem usage in Loop: 2288, per character mem: 108.952
charCount: 30, Running Memory free: 81856, mem usage in Loop: 3376, per character mem: 108.903
charCount: 40, Running Memory free: 80656, mem usage in Loop: 4576, per character mem: 111.61
charCount: 50, Running Memory free: 79536, mem usage in Loop: 5696, per character mem: 111.686
charCount: 60, Running Memory free: 78368, mem usage in Loop: 6864, per character mem: 112.525
charCount: 70, Running Memory free: 77232, mem usage in Loop: 8000, per character mem: 112.676
```

#### Sample text output from `TextMap` usage with Helvetica Bold 16:
```
After display.show(myGroup), just before loop start  Memory free: 86192
charCount: 10, Running Memory free: 86320, mem usage in Loop: -128, per character mem: -11.6364
charCount: 20, Running Memory free: 86192, mem usage in Loop: 0, per character mem: 0.0
charCount: 30, Running Memory free: 86192, mem usage in Loop: 0, per character mem: 0.0
charCount: 40, Running Memory free: 86192, mem usage in Loop: 0, per character mem: 0.0
charCount: 50, Running Memory free: 86192, mem usage in Loop: 0, per character mem: 0.0
```

#### Experimental setup:
- Itsy Bitsy NRF52840
- ILI9341 Display 320x240

#### Tabular Results:
[image]

#### Summary of results:
- For `label.py`, each added character increases memory usage by 110 to 125 bytes per character (dependent upon font size).
- For a large string (659 characters), the `label.py` has the largest overhead.
- Changing max_glyphs of 659 to 30 reduces the memory overhead by 5 kB, and then `label.py` has less overhead.

### Conclusion: Using a lot of text at ~100 bytes per character eats up memory in a hurry.

tannewt, 2020-07-29:
@kmatch98 Thank you for the empirical numbers!

I think the textmap numbers are a bit misleading though because it's a fixed overhead of 320 x 240 // 8 = 9600 bytes which means label is a better choice for strings of 9600 // 125 = 76 characters or less.

We can raise this number by reducing the overhead of each tilegrid. I think the cost of each character now is:
* 4 bytes for the Group entry pointer.
* the size of the `displayio_tilegrid_obj_t`
* the size of the glyph's bitmap

Only the first two bullets apply when a duplicate character is used.

I think we should add a `TextArea` class to this library (in a different file) that takes the approach you suggest. It can optimize for multi-line text areas (aka paragraphs) while label can focus on short text, usually in a single line.

kmatch98, 2020-07-29:
@tannewt To check your comment about the third bullet, reran the code with a string with just repeated "M" in it.

I see the same ~100 byte incremental memory usage when each character is added, even thought it is a duplicated glyph.  Please note that I load the glyphs before entering the loop:
```
glyphs = b'0123456789abcdefghijklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ-,.:?! '
font.load_glyphs(glyphs)`
```

Here is my code with the repeated "M" string using `label.py` and Helvetica 16.  Am I missing something?
```
code.py output:
Starting the display...
spi.frequency: 32000000
Display is started
loading fonts...
loading glyphs...
Glyphs are loaded.
Fonts completed loading.
After creating Group,  Memory free: 92048
myString: MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
string length: 154
After display.show(myGroup), just before loop start  Memory free: 89344
charCount: 10, Running Memory free: 88624, mem usage in Loop: 720, per character mem: 65.4545
charCount: 20, Running Memory free: 87280, mem usage in Loop: 2064, per character mem: 98.2857
charCount: 30, Running Memory free: 86048, mem usage in Loop: 3296, per character mem: 106.323
charCount: 40, Running Memory free: 84720, mem usage in Loop: 4624, per character mem: 112.78
charCount: 50, Running Memory free: 83456, mem usage in Loop: 5888, per character mem: 115.451
charCount: 60, Running Memory free: 82224, mem usage in Loop: 7120, per character mem: 116.721
charCount: 70, Running Memory free: 80896, mem usage in Loop: 8448, per character mem: 118.986
charCount: 80, Running Memory free: 79664, mem usage in Loop: 9680, per character mem: 119.506
charCount: 90, Running Memory free: 78400, mem usage in Loop: 10944, per character mem: 120.264
```

kmatch98, 2020-07-29:
To keep from getting lost in the empirical measurements:

Once we better define what the problem is, I want to make sure we have an agreed strategy before going any further.  

I see two major approaches:
1.  Make a drop-in replacement for `label.py`, only difference is that it's in a bitmap (and less mutable on the fly).  This is the suggestion that was discussed in the call on July 20th.
2.  Make a library that is good for rendering larger areas of text, possibly including other features (e.g. word wrapping, cursor control like a terminal)
3.  No change.

Based on what I have seen the current `label.py` uses an unexpectedly large amount of memory even for small amounts of text (> 76 chars?) and is causing people problems.  If that's not really a problem then path #2 is probably is a better approach, to focus on applications specifically that are text-heavy.  However, text-heavy applications may prefer "mutable" structures so we'll have to consider how to respond to that if path #2 is preferred.  

Y'all's feedback is welcome, requested and appreciated! @FoamyGuy @tannewt

tannewt, 2020-07-30:
> I see the same ~100 byte incremental memory usage when each character is added, even thought it is a duplicated glyph. Am I missing something?

Nope! I didn't realize you were preloading the glyphs.

> I see two approaches to consider:
> 
> 1. Make a drop-in replacement for `label.py`, only difference is that it's in a bitmap (and less mutable on the fly).  This is the suggestion that was discussed in the call on July 20th.
> 
> 2. Make a library that is good for rendering larger areas of text, possibly including other features (e.g. word wrapping, cursor control like a terminal)

I would do more 2 than 1. To me, a label is a small amount of text. Hundreds of characters is not a label but rather a paragraph or text area. I don't think it needs to be in a different library though. It should just be a different module (aka file) in this library.

> Based on what I have seen the current `label.py` uses an unexpectedly large amount of memory even for small amounts of text (> 76 chars?) and is causing people problems. If that's not really a [...]

I agree that 100 bytes per character is a lot. We should take a look at TileGrid to see if we can reduce that overhead. I don't think that is the primary concern of this investigation though because the memory errors people hit are due to large consecutive allocations because of the group containing many characters. 100 bytes per TG is a lot but at least they are separate memory chunks.

I'd propose adding a TextArea class that takes rectangular bounds and allocates a bitmap for the bounds once up front. Then, text can be laid out into it repeatedly without additional allocations.

> Y'all's feedback is welcome, requested and appreciated! @FoamyGuy @tannewt

Thank you for your in depth thoughts on this!

kmatch98, 2020-07-31:
Thanks @tannewt.  I remain suspicious of TileGrid, it seems like the overhead it excessive but I can't understand the code well enough to point to the issue.

After good discussion with @FoamyGuy this morning I decided to start with the direct drop-in replacement `bitmap_label` (type #1 above).  

Once that is done, I'll start on #2.  I've already been down this road with [TextMap](https://github.com/kmatch98/CircuitPython_textMap), so those features will be my starting scope.  Main scope is to decide what features to bring from recent updates to `label.py`.

But before I turn to #2 , I'd like elicit feedback on the top features desired.  I'm not sure the best forum to do that.  I suspect the easiest method is to create something and then people will say "I wish..."  The In-the-Weeds time is helpful, but I'm unsure if we can collect specific text-realted feedback there.  

I am totally new to this kind of large-scale collaborative development, so I'm open to your suggestions how to scope a new add-on file to the display_text library.

tannewt, 2020-07-31:
I think the first place to look for overhead is here: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/TileGrid.h#L37

Most of it could be simplified for the single tile case and we could therefore split much of it into a separate sub-allocation when more than one tile is used.

kmatch98, 2020-08-18:
Close with: https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/pull/79

closed 2020-08-18 by kmatch98