adafruit/Adafruit_CircuitPython_Display_Text #77: Use of get_bounding_box is not safe for certain fonts

kevinjwalters, 2020-08-03:
The documentation informs the user that there needs to be a **get_bounding_box** method on the font. There are fonts which do meet this criteria but may return `None`. Fonts loaded using `adafruit_bitmap_font` have a return value that depends on the _content_ of a `bdf` file:

https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font/blob/master/adafruit_bitmap_font/bdf.py#L67-L79

There are two sequence dereferences in the code currently for `[1]` from `get_bounding_box()` which won't handle this elegantly and will confuse users who are not intimately familiar with the content of the bdf file. Perhaps many/all bdf files include this data, though?

kmatch98, 2020-08-03:
According to the [BDF file specification](https://www.adobe.com/content/dam/acom/en/devnet/font/pdfs/5005.BDF_Spec.pdf), This parameter `FONTBOUNDINGBOX` is required, see snippet below.  Should this raise an exception if the item is missing?

I agree that it’s a bit obtuse to users about what `get_bounding_box` means.  Maybe these item names could be refigured with slighty descriptive names, maybe something like:

```
font.glyph_max_x = font. get_bounding_box[0]
font.glyph_max_y = font.get_bounding_box[1]
font.glyph_max_x_offset = font.get_bounding_box[2]
font.glyph_max_y_offset = font.get_bounding_box[3]
```
 
[image]

kevinjwalters, 2020-08-03:
Perhaps this should be punted over to https://github.com/adafruit/Adafruit_CircuitPython_Bitmap_Font then?