adafruit/Adafruit_CircuitPython_Display_Text #78: `y_offset` should not include any impact due to newlines '\n'

kmatch98, 2020-08-04:
The `y_offset` is currently incorrectly calculated with the inclusion of a multiple related to the quantity of newlines '\n' in the text string.

However `y_offset` should be independent of the number of lines in the string, and should just be related to the glyph dimensions, since this acts as a baseline offset to the vertical center of the "M" glyph.  

I observed this issue when calibrating text to the upcoming `bitmap_label.py` file.

dherrada, 2020-08-06:
@kmatch98 Good find. If you're up for it, a PR would be greatly appreciated.

kmatch98, 2020-08-07:
Thanks @dherrada, please put this into my queue and I will figure it out.  Also please note there will be a similar issue to resolve in the parallel library `bitmap_label.py`.