adafruit/Adafruit_CircuitPython_Display_Text #80: y position of text can be wrong if constructor does not specify text

kevinjwalters, 2020-08-07:
I updated my libraries last night to ones from `adafruit-circuitpython-bundle-5.x-mpy-20200806.zip`. I noticed the performance improvement with moving around text but I also spotted one solitary piece of text that was misplaced slightly off screen.

I've just made a simple recreation of this, I think it's when `text` is not specified or it is empty string, `y` is set, and then `text` is set. I am using scaled text here.

```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.3.0 on 2020-04-29; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> import displayio, terminalio, board
>>> from adafruit_display_text.label import Label
>>> fdim = terminalio.FONT.get_bounding_box()
>>> l1 = Label(terminalio.FONT, max_glyphs=9, scale=3, color=0xff0000)
>>> l2 = Label(terminalio.FONT, text="", max_glyphs=9, scale=3, color=0x00ff00)
>>> l3 = Label(terminalio.FONT, text="Chad", max_glyphs=9, scale=3, color=0x0000ff)
>>> l1.y = l2.y = l3.y = round(240 - (3 * fdim[1] / 2))
>>> (l1.y, l2.y, l3.y)
(219, 219, 219)
>>> l2.x = 80
>>> l3.x = 160
>>> tg = displayio.Group(max_size=3)
>>> tg.append(l1)
>>> tg.append(l2)
>>> tg.append(l3)
>>> board.DISPLAY.show(tg)
>>> l1.text="Wot"
>>> l2.text="No"
>>> (l1.y, l2.y, l3.y)
(240, 240, 219)
```

Only `l3` is correctly position, `l1` and `l2` are half off screen.

I see #78 which sounds like it could be in the same area?

kmatch98, 2020-08-07:
Put this in my queue.  I think it’s probably different than the case you mention. My first thought is that the `bounding_box` has something wrong with it but I’ll have to dig in. Glad you found a workaround in the meantime.

kmatch98, 2020-08-08:
@kevinjwalters thanks for the heads-up on this issue with the `text` setter.  

I uncovered that the `anchor_point` is being set to (0,0) as a default value.  So, when you update the text, it checks the anchor_point and anchored_position as if they are valid data.

The code should be updated such that `anchor_point = None` when the anchor_point isn't defined.  

There will need to be associated changes with the anything that uses the `anchor_point` or get/sets the `anchored_position` to respond properly when `anchor_point = None`.  It'll take me a day or two to sort through this and test it out a bit.

kevinjwalters, 2020-08-10:
I re-ordered by code as I can easily set the `text` to its (final) value when the `Label` is constructed.

kmatch98, 2020-08-12:
This was resolved by the merge PR #81

FoamyGuy, 2020-08-15:
Closed by #81 thanks @kmatch98

closed 2020-08-15 by FoamyGuy