adafruit/Adafruit_CircuitPython_Display_Text #91: Empty text string causes a Label's y-attribute to change

CedarGroveStudios, 2020-08-27:
After a single character label with an empty text-attribute is defined, the label's y-attribute is incremented by 7 when the label's text-attribute is loaded with a character (TEST 1). Conversely, if the label is defined with a character in the text-attribute, the y-attribute will decrement by 7 when the text-attributed is emptied (TEST 3). In either case, the y-attribute shouldn't change. Looks like something associated with the anchor point calculation updating the y-attribute.

Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit ItsyBitsy M4 Express with samd51g19
CircuitPython 5.x Library Bundle 20210-08-26

Test code:
```import terminalio
from   adafruit_display_text.label   import Label

font_0 = terminalio.FONT  # Internal font

print('CircuitPython 5.x Library Bundle 20210-08-26')
address = (12, 14)
print('adress:', address)

print()
print('TEST 0: pre-load Label text, change text later')
test_0 = Label(font_0, text='0', color=0xFFFFFF, max_glyphs=1)
test_0.x, test_0.y = address
print('x =', test_0.x, 'y=', test_0.y)
test_0.text = '1'
print('x =', test_0.x, 'y=', test_0.y)

print()
print('TEST 1: no Label text, change text later')
test_1 = Label(font_0, text='', color=0xFFFFFF, max_glyphs=1)
test_1.x, test_1.y = address
print('x =', test_1.x, 'y=', test_1.y)
test_1.text = '1'
print('x =', test_1.x, 'y=', test_1.y)

print()
print('TEST 2: no Label text, no text in change')
test_2 = Label(font_0, text='', color=0xFFFFFF, max_glyphs=1)
test_2.x, test_2.y = address
print('x =', test_2.x, 'y=', test_2.y)
test_2.text = ''
print('x =', test_2.x, 'y=', test_2.y)

print()
print('TEST 3: pre-load Label text, no text in change')
test_3 = Label(font_0, text='0', color=0xFFFFFF, max_glyphs=1)
test_3.x, test_3.y = address
print('x =', test_3.x, 'y=', test_3.y)
test_3.text = ''
print('x =', test_3.x, 'y=', test_3.y)
```

REPL output: 

```Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit ItsyBitsy M4 Express with samd51g19
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
CircuitPython 5.x Library Bundle 20210-08-26
adress: (12, 14)

TEST 0: pre-load Label text, change text later
x = 12 y= 14
x = 12 y= 14

TEST 1: no Label text, change text later
x = 12 y= 14
x = 12 y= 21

TEST 2: no Label text, no text in change
x = 12 y= 14
x = 12 y= 14

TEST 3: pre-load Label text, no text in change
x = 12 y= 14
x = 12 y= 7

Press any key to enter the REPL. Use CTRL-D to reload.```

kmatch98, 2020-08-27:
Thanks for the detailed description of the issue!  

As you suggested, it’s likely an issue with the `anchored_position` or maybe the `bounding_box` calculation of a blank string.  I will check it out and get back with you.

kmatch98, 2020-08-27:
@CedarGroveStudios I tested using the latest version and it works ok for me.  I think you are using a slightly older version of the `label`. 

**Can you please pull the latest `adafruit_display_text\label` library and let me know your results?**  [Here is the circuitpython.org page with the latest library bundle.](https://circuitpython.org/libraries)

Here's what I found:
1. I used the 8/26 library bundle that you specified and it worked correctly for me.   It worked both for `label.mpy` from the bundle and `label.py` from the latest library [here](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/tree/master/adafruit_display_text).
1. I went back to the 8/11 version of the library bundle and the results match what you showed above.  Here is the [older library version I used with the label.py file.](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/tree/ab0158d62395fb9dbe49a684829d6468a70ace86/adafruit_display_text)

### 1. Results with the 8/26 library bundle - I think this is working right
Here is the output I get from running your code on the latest `label.py` library.  I think this is the result that you should get.
```
adress: (12, 14)

TEST 0: pre-load Label text, change text later
x = 12 y= 14
x = 12 y= 14

TEST 1: no Label text, change text later
x = 12 y= 14
x = 12 y= 14

TEST 2: no Label text, no text in change
x = 12 y= 14
x = 12 y= 14

TEST 3: pre-load Label text, no text in change
x = 12 y= 14
x = 12 y= 14
```

### 2. Results with the 8/11 library bundle specified above - This matches the results you showed
```
adress: (12, 14)

TEST 0: pre-load Label text, change text later
x = 12 y= 14
x = 12 y= 14

TEST 1: no Label text, change text later
x = 12 y= 14
x = 12 y= 21

TEST 2: no Label text, no text in change
x = 12 y= 14
x = 12 y= 14

TEST 3: pre-load Label text, no text in change
x = 12 y= 14
x = 12 y= 7
```

### The test code I used
Here is the code that I ran.  The only change I made was `import terminalio` at the top:
```
import terminalio
from adafruit_display_text.label import Label

font_0 = terminalio.FONT  # Internal font

print('CircuitPython 5.x Library Bundle 20210-08-26')
address = (12, 14)
print('adress:', address)

print()
print('TEST 0: pre-load Label text, change text later')
test_0 = Label(font_0, text='0', color=0xFFFFFF, max_glyphs=1)
test_0.x, test_0.y = address
print('x =', test_0.x, 'y=', test_0.y)
test_0.text = '1'
print('x =', test_0.x, 'y=', test_0.y)

print()
print('TEST 1: no Label text, change text later')
test_1 = Label(font_0, text='', color=0xFFFFFF, max_glyphs=1)
test_1.x, test_1.y = address
print('x =', test_1.x, 'y=', test_1.y)
test_1.text = '1'
print('x =', test_1.x, 'y=', test_1.y)

print()
print('TEST 2: no Label text, no text in change')
test_2 = Label(font_0, text='', color=0xFFFFFF, max_glyphs=1)
test_2.x, test_2.y = address
print('x =', test_2.x, 'y=', test_2.y)
test_2.text = ''
print('x =', test_2.x, 'y=', test_2.y)

print()
print('TEST 3: pre-load Label text, no text in change')
test_3 = Label(font_0, text='0', color=0xFFFFFF, max_glyphs=1)
test_3.x, test_3.y = address
print('x =', test_3.x, 'y=', test_3.y)
test_3.text = ''
print('x =', test_3.x, 'y=', test_3.y)

while True:
    pass
```

CedarGroveStudios, 2020-08-27:
Thank you for your speedy response @kmatch98 ! I was able to confirm that 8/26 bundle release had indeed provided the fix. My _always reliable_ automatic library update routine skipped over `adafruit_display_text` and a couple of other libraries during its latest execution. Sorry about the fire drill.

closed 2020-08-27 by CedarGroveStudios

kmatch98, 2020-08-27:
No worries, I’m just glad we pre-emptively solved the issue.  Cheers!