adafruit/Adafruit_CircuitPython_Display_Text #94: Multiline text is no longer vertically centred around the all of the text lines

kevinjwalters, 2020-10-20:
I just tested the CLUE sensor plotter with latest libraries from the 20201020 bundle and noticed its button guide screen has the text incorrectly positioned. You can see how it should look from the image at the top of https://learn.adafruit.com/clue-sensor-plotter-circuitpython/circuitpython-sensor-plotter but now it appears with the first line saying `Button Guide` half way down the screen.

I think this code reproduces the problem. `Line 1` initially appears in the middle of the screen but then stays exactly where it is at lines are added. Judging by the plotter code this wasn't the behaviour of the original library with _multi-line_ text.

```
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> import displayio
>>> import terminalio
>>> import board
>>> import time
>>> from adafruit_display_text.label import Label
>>> text = Label(terminalio.FONT, text="line 1", max_glyphs=200)
>>> board.DISPLAY.show(text)
>>> text.y = board.DISPLAY.height // 2
>>> for lines in range(2, 13 + 1):
...     text.text = "\n".join(["Line " + str(line) for line in range(1, lines + 1)])
...     time.sleep(2)
...
...
...
```

FoamyGuy, 2020-10-20:
I'm not sure when specifically that this behavior got changed, but I am leaning toward thinking the current behavior is more correct to be honest. 

This code only sets the `y` position once in this line `text.y = board.DISPLAY.height // 2` which is setting the y position to midway down the screen vertically. 

Since this code is not using Anchored position I don't think any automatic centering (in any orientation) should be taking place really. It should be staying in the same location based on `x,y` positioning based on the default x/y origin.

In the older version of the library the y position appears to keep changing even though the user code is not making any more changes to it, which I think can be confusion to users.

FoamyGuy, 2020-10-20:
I modified your code to use anchored position and this way it behaves more like it did in the older version of the library even when I switch back to the new one: 
```python
import displayio
import terminalio
import board
import time
from adafruit_display_text.label import Label
text = Label(terminalio.FONT, text="line 1", max_glyphs=200)
board.DISPLAY.show(text)
#text.y = board.DISPLAY.height // 2

text.anchor_point = (0, 0.5)
text.anchored_position = (4, board.DISPLAY.height // 2)
for lines in range(2, 13 + 1):
    text.text = "\n".join(["Line " + str(line) for line in range(1, lines + 1)])
    time.sleep(2)
```

FoamyGuy, 2021-01-09:
I'm going to close this one. We can re-open if you anyone wants to revisit the x,y positioning specifically.

closed 2021-01-09 by FoamyGuy