adafruit/Adafruit_CircuitPython_Display_Text #95: Error in examples/display_text_simpletest.py

lastcoolnameleft, 2020-11-06:
I have the recent Adabox016 and when I try to use the simpletest.py, I get:
```
Traceback (most recent call last):
  File "code.py", line 10, in <module>
AttributeError: 'module' object has no attribute 'DISPLAY'
```

I'm using the library from adafruit-circuitpython-bundle-6.x-mpy-20201030

lastcoolnameleft, 2020-11-06:
I'm experiencing this issue with `adafruit-circuitpython-matrixportal_m4-en_US-6.0.0-rc.1.uf2`

FoamyGuy, 2020-11-06:
The example code is assuming that it will get run on a device that has a display built in (in which case it would be available as `board.DISPLAY` within the code). 

But the Matrix Portal needs to do a bit of setup first before it can start using the display. 

There is a [MatrixPortal library](https://github.com/adafruit/Adafruit_CircuitPython_MatrixPortal) which will do lots of the set up for you. If you use that the simpletest example code can be adapted to work on the Matrix portal like this:
```python
import terminalio
from adafruit_display_text import label
from adafruit_matrixportal.matrix import Matrix
matrix = Matrix()
display = matrix.display

text = "Hello\nworld"
text_area = label.Label(terminalio.FONT, text=text)
text_area.x = 1
text_area.y = 4
display.show(text_area)
while True:
    pass
```

FoamyGuy, 2020-11-06:
@lastcoolnameleft This example has been added in the examples directory now: https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/blob/master/examples/display_text_matrixportal.py

I'm going to close this issue for now. But if you end up having trouble when you try this feel free to ping me here or in the discord I can help you out.

closed 2020-11-06 by FoamyGuy

lastcoolnameleft, 2020-11-07:
Thanks for the update and clarification about how/when to use it.  The documentation just pointed me to the repo so I was unaware that the code was for a specific board.  I'm good with it being closed.