adafruit/Adafruit_CircuitPython_Display_Text #97: bitmap_label

FoamyGuy, 2020-11-14:
bitmap_label will not draw text if it's set after creation instead of passing `text` in the constructor
```python
import board
import terminalio
from adafruit_display_text import bitmap_label
from adafruit_display_text import label

text = "Hello world"
# text_area = label.Label(terminalio.FONT, max_glyphs=50)  # regular label does show the text afterward.

text_area = bitmap_label.Label(terminalio.FONT)
# text_area = bitmap_label.Label(terminalio.FONT, text=" ") # if we init with a space it will show the text
text_area.text = text
text_area.x = 10
text_area.y = 10
board.DISPLAY.show(text_area)
while True:
    pass
```
I can take a look into making the bitmap_label work the same as label in this scenario. Also am curious if @kmatch98 have any ideas if you have a chance to take a look.

ladyada, 2020-11-14:
what board is this, magtag? you have to force a refresh if so

FoamyGuy, 2020-11-14:
@ladyada  PyPortal and UM Feather S2 with 128x32 OLED. Still waiting for delivery on magtag

kmatch98, 2020-11-14:
Let me try it on the PyPortal and see what’s going on.

kmatch98, 2020-11-14:
This was a bug that was introduced when the separate `self` and `self.local_group` was added to handle modifying the scaling on the fly.

The bug was when the text was "" and it actually was modifying the `self` group rather than the `self.local_group`.  Pull request is incoming.

FoamyGuy, 2020-11-14:
resolved by #98

closed 2020-11-14 by FoamyGuy