adafruit/Adafruit_CircuitPython_Display_Text #108: Setting anchor_point and anchored_position in constructor broken for bitmap_label

FoamyGuy, 2021-01-17:
fails with this error message:
```
Traceback (most recent call last):
  File "code.py", line 136, in <module>
  File "/lib/adafruit_display_text/bitmap_label.py", line 146, in __init__
  File "/lib/adafruit_display_text/bitmap_label.py", line 201, in _reset_text
  File "/lib/adafruit_display_text/bitmap_label.py", line 656, in anchor_point
  File "/lib/adafruit_display_text/bitmap_label.py", line 672, in anchored_position
AttributeError: 'Label' object has no attribute '_bounding_box'
```

label is unaffected by this issue.

kmatch98, 2021-01-17:
@FoamyGuy Thanks for bringing this up, this was a bug by me.  

Also, somehow I'm not sure my library was up to date, but I found an issue when the bitmap_label was instanced with an empty text string.  I thought this was fixed before.  I thought updated my library to the latest, but the error was still in there.  Strange.  

If you see something wrong with the commit history, let me know and I will consult with the git gurus to figure out how to resolve.

I'd appreciate if you verify this fixes the issue you observed.

FoamyGuy, 2021-01-18:
closed by #109

closed 2021-01-18 by FoamyGuy