adafruit/Adafruit_CircuitPython_Display_Text #113: The scale attribute on Label seems to be being applied twice.

lesamouraipourpre, 2021-02-06:
When setting the scale of text created with Label, the scale appears to be applied twice causing  the scale to be squared.

In the test script below the word 'Scale' is printed to screen in scale=1, 2 and 3
[text_scale_test.py.txt](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/files/5937194/text_scale_test.py.txt)

The output txt follows, note the scale in the absolute_transform:
[text_test_output.txt](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/files/5937193/text_test_output.txt)

Below is a photograph of the resulting image on my Pimoroni 128x128 OLED:
[image]

FoamyGuy, 2021-02-06:
I think this bug is specifically inside of Blinka_Displayio instead of this library. If you there is an issue here that is for what I think the root cause is: https://github.com/adafruit/Adafruit_Blinka_Displayio/issues/45

lesamouraipourpre, 2021-02-06:
After a few hours of trying to figure out the design of Label, it seems to be because of the sub-Group within the label, the __init__ method refers to 
```        # instance the Group
        # self Group will contain a single local_group which contains a Group (self.local_group)
        # which contains a TileGrid
        # The self scale should always be 1
        super().__init__(max_size=1, scale=1, **kwargs)
        # local_group will set the scale
        self.local_group = displayio.Group(max_size=max_glyphs + 1, scale=scale)
```

However, both the local scale and the local_group scale value are adjusted when scale is updated after a transformation
```    @scale.setter
    def scale(self, new_scale):
        current_anchored_position = self.anchored_position
        self._scale = new_scale
        self.local_group.scale = new_scale
        self.anchored_position = current_anchored_position
```
The attached diff is one possible fix, that keeps the local scale at 1 and uses the scale from the local_group
[diff.txt](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/files/5937774/diff.txt)

This produces the attached image:
[image]

kmatch98, 2021-02-07:
@lesamouraipourpre 
Good find on this bug. I agree with your solution. Can you submit a pull request to fix?

FoamyGuy, 2021-02-07:
Nice! I tested this out successfully with Blinka_Displayio and it does resolve the extra scaling. Thank you for digging into this and finding a solution @lesamouraipourpre.

I am still kind of curious about what the difference between core displayio and blinka_displayio that might cause the current code to behave differently.

FoamyGuy, 2021-02-11:
resolved by #116 and #117

closed 2021-02-11 by FoamyGuy