adafruit/Adafruit_CircuitPython_Display_Text #118: Tabs are ignored in `label` and `bitmap_label`

kmatch98, 2021-02-11:
Currently, all tabs `'\n'` are ignored in both label and bitmap_label.

lesamouraipourpre, 2021-02-12:
That could be a challenge to define what a tab is. Do you define it as a fixed number of pixels, eg 25px, or do you define it based on the width of a character, eg 8 spaces. As a programmer, I would go with 4 spaces.

Also, how would it behave if the text is not left anchored?

jposada202020, 2021-02-12:
Agree with lesamouraipourpre, at the end of the day the 4 spaces will get translated to pixels in the library according to the Font used

jposada202020, 2021-02-19:
@kmatch98 @FoamyGuy if you are not currently wrking at this I can take a look tonight. Let me know. :)

FoamyGuy, 2021-02-19:
@jposada202020 I'd say go for it. Thank you! I'm going to test out the bitmap_label alignment PR next when I get back to it. This was a bit lower down the list for me, so am happy to have you take it on if you are willing to.