adafruit/Adafruit_CircuitPython_Display_Text #125: Duplicate code in library files - potential refactor needed

FoamyGuy, 2021-03-04:
actions failure with details on duplicated sections is [here](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/runs/2017909871?check_suite_focus=true#step:11:31)

We want to avoid having duplicated code if possible. One way to avoid it here will be to make a `LabelBase` class in `__init__.py` with the duplicated parts and extend that in the other types of labels.

jposada202020, 2021-03-05:
I I took a look at this yesterday. For simple modules, I had success creating  LabelBase, in the __init__ file, then creating new classes the label and bitmap_label as children of display.Group and LabelBase. The setter and the getter needs to be equal in order to use them, however this could be solved using a belonging flag and change the module behavior. However, for some modules although equal, ie background (color and calculation) the way both libraries do some calculations, is different, and calling methods inside them also, complicating things inside LabelBase . A first glance I did not see any "easy" path for this

I see as the possible `easy` to share in the BaseLabel the following modules
* anchored_position
* background_color
* _get_ascent_descent

Bitmap_label is known to save memory, we would need to keep this feature in doing this change as some people have bitmap_label in their projects to save memory.

FoamyGuy, 2021-03-05:
I will work on this some tonight or tomorrow as well.

Unfortunately I think we may need to get this sorted before we can merge the other PRs because this will cause actions to fail currently.

jposada202020, 2021-03-05:
@FoamyGuy thanks, Agree, we will work on this, let me know if you want that we do a little brainstorm session.

FoamyGuy, 2021-03-05:
@jposada202020 I'm going to be working on this some tonight starting in a little bit. If you are around and still interested in brainstorm session ping me on Discord.

jposada202020, 2021-03-06:
@FoamyGuy working in the turning text. Enclosed is a prototype that uses the internal machine of label

https://github.com/jposada202020/sandbox_display_text/blob/0e4257dde2899ad269f30510668780d66536275f/developing_turning_text.py

[image]

will wait in your findings..

PR #126 (2021-03-09): Label base class
Resolves #125 

LabelBase class has been introduced inside of __init__.py and bitmap_label and label now extend LabelBase which removes all of the duplicated code from them.

I've tested successfully on Blinka_Displayio, PyPortal, and MagTag. This is a relatively large refactor though, happy to get any additional testing from other folks.

One thing I was unsure about was the sphinx docstrings. I think we want to make sure all of the relevant parameters will be visible in the subclass docs pages. I don't know if we should have the parameters documented in just the base, or jus [... cut]
closed 2021-03-09 by FoamyGuy