adafruit/Adafruit_CircuitPython_Display_Text #131: Update label to use new displayio.Group

FoamyGuy, 2021-03-08:
Making an issue now so that we keep this in mind. I do think we'll want to hold off on making this change at least until there is a stable release made that includes the changes to Group though so it might be a little bit until we actually need to do anything further.

displayio.Group has been updated to allow it's size to be mutable so more things can be added, and there is no need to use the `max_size` any longer.

We will be able to change label.Label to make use of this and remove the `max_glyphs` restriction.

tannewt, 2021-03-08:
Let's start moving things off max_size once 6.2.0 has been stable for a little while. We don't want to drop max_size with 6.1.x

FoamyGuy, 2021-07-01:
I'm going to re-open this one for now with the intent to figure out if we can safely remove `max_glyphs` from label. If it turns out not to be possible we can close this back up.

PR #153 (2021-07-01): Remove max_size parameter
Closes #131

Remove the `max_size` parameter from `Label` and `BitmapLabel` now that it has been disabled in `shared-bindings`.

This has been tested with:
```
Adafruit CircuitPython 6.3.0 on 2021-06-01; Adafruit PyPortal with samd51j20
```
commit 3f9a6b6c (2021-07-16): Refactor max_glyphs away (Closes #131)
closed 2021-07-16 by FoamyGuy