adafruit/Adafruit_CircuitPython_Display_Text #136: When using backgroung setter, background text does not shows correctly

jposada202020, 2021-03-10:
When ussing the setter for background text, the text does not show correctly. I am assuming that this applies for anchor point setter, as I was receiving complains from the code that I was testing. However I thought it was my new PR. Did not test this for anchor_point. 

So sorry for not to catching this before the Refactor, but normally I do all the test with list iterations. sorry @FoamyGuy . 

## Test Code
```python
import terminalio
import displayio
from os import uname
if uname()[0] == 'samd51':
    import board
else:
    from blinka_displayio_pygamedisplay import PyGameDisplay
from adafruit_display_text import label
from adafruit_bitmap_font import bitmap_font

if uname()[0] == 'samd51':
    display= board.DISPLAY
else:
    display = PyGameDisplay(width=320, height=240)
splash = displayio.Group(max_size=10)
MEDIUM_FONT = bitmap_font.load_font("fonts/LibreBodoniv2002-Bold-10.bdf")

bitmap = displayio.Bitmap(4, 320, 2)
palette = displayio.Palette(2)
palette[0] = 0x004400
palette[1] = 0x00FFFF
tile_grid = displayio.TileGrid(bitmap,
                               pixel_shader=palette,
                               x=155,
                               y=0)
splash.append(tile_grid)

bitmap = displayio.Bitmap(320, 4, 2)
tile_grid = displayio.TileGrid(bitmap,
                               pixel_shader=palette,
                               x=0,
                               y=110)
splash.append(tile_grid)

text = "CircuitPython"
text_area = label.Label(MEDIUM_FONT,
                        text=text,
                        x=155,
                        y=155,
                        padding_left=10,
                        padding_top=10,
                        padding_bottom=10,
                        background_color = 0x990099,
                        padding_right=10)

splash.append(text_area)

text = "CircuitPython"
text_area = label.Label(MEDIUM_FONT,
                        text=text,
                        background_tight=True,
                        padding_left=10,
                        padding_top=10,
                        padding_bottom=10,
                        padding_right=10)
text_area.x = 155
text_area.y = 110
text_area.background_color = 0x990099
splash.append(text_area)

display.show(splash)

```

## Results
[image]

jposada202020, 2021-03-10:
I have tested in Release 2.15.0 and works as expected, using the same code included

FoamyGuy, 2021-03-10:
Good catch, I am glad you found this! I think this is the same issue from #49. I must have messed up the handling of background bitmap in refactoring. I will look into this tonight and get a fix in.

PR #137 (2021-03-11): Fix label background_color setter
This resolves #136 

Setting the background_color previously worked differently in BitmapLabel vs. Label. They were inadvertently combined during the refactor to behave the same  which caused trouble for Label.

These changes break it out into separate implementations so that each can have their own correctly working behavior.

I tested with:
```
Adafruit CircuitPython 6.2.0-beta.3-36-g103632f44 on 2021-03-07; Adafruit PyPortal with samd51j20
```
using the code from the error as well as a few other background color example scripts from this repo.  Everything s [... cut]
closed 2021-03-11 by FoamyGuy