adafruit/Adafruit_CircuitPython_Display_Text #140: anchored position issue

FoamyGuy, 2021-03-13:
Issue and test code outlined here: https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/pull/137#issuecomment-795956051

jposada202020, 2021-03-20:
@FoamyGuy I have retested this, with the new refactor. This is working as expected. There is A mistake in my test code.  I test this in the pyportal and in the Pygame Display without any problems
The problem was in the setter as I was not using the `=` sign. If you coudl retest  it and verify that everything is working
```python

import terminalio
import displayio
from os import uname
import time

if uname()[0] == 'samd51':
    import board
else:
    from blinka_displayio_pygamedisplay import PyGameDisplay
from adafruit_display_text import label, bitmap_label
from adafruit_bitmap_font import bitmap_font

if uname()[0] == 'samd51':
    display = board.DISPLAY
else:
    display = PyGameDisplay(width=320, height=240)
splash = displayio.Group(max_size=10)
MEDIUM_FONT = bitmap_font.load_font("fonts/Helvetica-Bold-16.bdf")
LINE_HOR = displayio.Bitmap(4, 320, 2)
LINE_VER = displayio.Bitmap(320, 4, 2)
palette = displayio.Palette(2)
palette[0] = 0x004400
palette[1] = 0x00FFFF

text = "CircuitPython"
text_area = bitmap_label.Label(MEDIUM_FONT,
                        text="First Text",
                        label_direction="LTR",
                        background_tight=True,
                        x=200,
                        y=100,
                        padding_left=10,
                        padding_top=10,
                        padding_bottom=10,
                        padding_right=10,

                               )
text_area.anchored_position=(200, 100)
text_area.anchor_point=(0.5, 0.5)
splash.append(text_area)

text_area_UPR = bitmap_label.Label(MEDIUM_FONT, text=text,
                        # label_direction="UPR",
                        x=10,
                        y=10,)

splash.append(text_area_UPR)

text_area_DWR = bitmap_label.Label(MEDIUM_FONT,
                        text="Upwards",
                        label_direction="UPR",
                        # x=200,
                        # y=200,
                        anchored_position=(200, 200),
                        anchor_point=(0.5, 1.0),

)

line_ver= displayio.TileGrid(LINE_VER,
                               pixel_shader=palette,
                               x=0,
                               y=200)
splash.append(line_ver)
line_hor= displayio.TileGrid(LINE_HOR,
                               pixel_shader=palette,
                               x=200,
                               y=0)
splash.append(line_hor)

splash.append(text_area_DWR)
display.show(splash)

while True:
    pass

```

FoamyGuy, 2021-03-31:
I confirmed the anchor positioning is working as intended with the given script and a modified simpletest. 

Thanks for looking into this as well @jposada202020

closed 2021-03-31 by FoamyGuy