adafruit/Adafruit_CircuitPython_Display_Text #155: Width property not working in some cases

FoamyGuy, 2021-07-24:
Ref: https://github.com/adafruit/Adafruit_CircuitPython_MacroPad/issues/10

Width and height properties do exist in LabelBase. I'm not sure why it's raising AttributeNotFound in this context. 

I am looking into it further and working on a fix now.

FoamyGuy, 2021-07-24:
I tested this minimal example:
```python
import board
import terminalio
from adafruit_display_text import bitmap_label

text = "Hello world"
text_area = bitmap_label.Label(terminalio.FONT, text=text)
text_area.x = 10
text_area.y = 10
board.DISPLAY.show(text_area)

print(text_area.width)
while True:
    pass

```
It does print the width and does not raise an AttributeError. So this is somehow tied into the usage inside of grid_layout. Trying that out next to see if I can find a minimal reproducer.

lesamouraipourpre, 2021-07-24:
The way I did the width and height properties on `LabelBase`, they are read-only. That might be the source of the error.

Investigating further as well but I don't have a MacroPad, so I'll have to knock up a similar but different test.

FoamyGuy, 2021-07-24:
> The way I did the width and height properties on `LabelBase`, they are read-only. That might be the source of the error.
> 
> Investigating further as well but I don't have a MacroPad, so I'll have to knock up a similar but different test.

Yep, I think you are right. GridLayout is trying to set them and that is causing the error. In the old version they existed as class variables (unsure if proper term) which allowed them to be set even though it did not have a real effect.

I think the best solution for now will be to catch the error in GridLayout. I'm going to try that out on the stream today.

FoamyGuy, 2021-07-24:
It turned out this needed to be fixed inside the DisplayIO_Layout library. The PR to resolve this is here: https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/pull/42

closed 2021-07-24 by FoamyGuy