adafruit/Adafruit_CircuitPython_Display_Text #171: Suppress warnings in bitmap_label

Neradoc, 2022-05-12:
There are warnings printed in bitmap_label (but not label), if a glyph is missing, and if a glyph is clipped.
Those generate possibly unwanted "noise" to the REPL, it would be nice to silence them.

But I don't know if we'll want to: remove them or have an option to silence them or an option to display them, and if so, we likely want it to work on both types of labels.

jposada202020, 2023-02-15:
I think that we could do something similar to this 

https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/blob/64b75e12e56d5518aa662dce35711e36ce031856/adafruit_displayio_layout/widgets/cartesian.py#L80

Normally you would not find any trouble with the glyphs as the ascender/descender will be pretty standard across the letters , there are some exceptions fonts , and with the work done in https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/issues/103 it was helpful for debugging/information purposes.

Depending on the font, there could be glyphs that are not present, this is more prevalent for not common characters, so it is to inform the user that the character that they want to display is not present in the font file.

all that to say that I am in favor of using the Verbose argument :)

FoamyGuy, 2024-11-14:
I think this was resolved by #181 with the verbose flag which now provides a way to opt in or out of the extra messages.

closed 2024-11-14 by FoamyGuy