adafruit/Adafruit_CircuitPython_Display_Text #182: Unable to update scrolling_label text outside constructor

VinceVega7, 2023-02-24:
Hi,

It seems that you can not update the text of a scrolling_label outside the constructor. 
If outside the constructor but before the showing the display, if I update it to "test" it still retains "Hello World"

eg.

```
import displayio
import terminalio
import time
from adafruit_matrixportal.matrix import Matrix
from adafruit_display_text.scrolling_label import ScrollingLabel

matrix = Matrix()
display = matrix.display

# setup the label
font = terminalio.FONT
splash = displayio.Group()

title_label = ScrollingLabel(
    text="Hello World", font=font, color=0xFFFFFF, scale=1,
    anchored_position=(0, 0), anchor_point=(0, 0),
)

#### Here this does not work
title_label.text = "test"

splash.append(title_label)
display.show(splash)

while True:

   title_label.update()
```

jposada202020, 2023-02-25:
Hi :), you could use 

``` python

title_label.full_text = "Changing the Text"

```
Hoping this is what you need!

FoamyGuy, 2023-03-13:
Jposada is correct about `full_text` being the intended way to update the text on a ScrollingLabel.

However it was written that way initially moreso because I didn't know how to correctly override the parent property and still make the rest of the code work properly. 

After thinking on it a bit I do think it would be ideal if `text` property could be used though because just like @VinceVega7 was expecting I think average user is going to assume it will change out the entire text since that is how it behaves for the other label types. 

I've created #188 to override it so that the API functionality now matches more closely.

PR #188 (2023-03-13): override text property from parent on ScrollingLabel
This resolves #182 

As was noted on a comment in the issue `full_text` was the property that was used to change the text rather than `text` like it is with BitmapLabel and Label. So the functionality to change text after creation does exist, but the property name used to do it is not the same as it is on the other Label types.

This was original done that way because I didn't know how to override parent properties while still being able to use the parent implementation of the property in some other parts of code. It turned out that we skirt [... cut]
closed 2023-03-13 by jposada202020