adafruit/Adafruit_CircuitPython_Display_Text #200: Program stops without any error when importing the library

mirkokral, 2023-11-27:
Hello, my program stops without any output while importing the library, and i can't get it to work.
maybe helpful info: the program is ran using exec()
Code:
```python
print(1)
import board, digitalio
print(2)
from adafruit_display_text import bitmap_label as label
print(3)
from adafruit_bitmap_font import bitmap_font
```
(print statements only for debugging)
Serial output:
```
1
2
```
And no three. Can somebody help? (this happens in both bitmap label and normal label.

FoamyGuy, 2023-11-27:
@mirkokral What device are you running this code on? A circuitpython microcontroller? Or on a PC or raspberry pi in "normal" python? 

I'm not sure what you mean when you said "the program is ran using exec()" either can you elaborate any further about how you are executing this code? On a microcontroller it should just be saved as code.py in order to be run automatically by the device. On a PC or Raspberry pi under CPython I think it should be run with a command like `python my_script.py` and the code would be contained inside of that file that you specify.

mirkokral, 2023-11-28:
> @mirkokral What device are you running this code on? A circuitpython microcontroller? Or on a PC or raspberry pi in "normal" python?
>
> I'm not sure what you mean when you said "the program is ran [...]

The board is a wio terminal, i updated the post.

FoamyGuy, 2023-12-06:
@mirkokral are you saving your file as code.py on the CIRCUITPY drive? or doing something different?

mirkokral, 2024-04-14:
Old issue and also it was something with my sd card (because i use an sd card to run my code)

closed 2024-04-14 by mirkokral