adafruit/Adafruit_CircuitPython_Display_Text #202: Add error handling for misspelled kwargs in adafruit_display_text.label

bradanlane, 2024-02-12:
The parameters for the `adafruit_display_text.label` are **anchor_point** and **anchored_position**. This is confusing and results in frequent user errors. The error is not caught by Python which is difficult for a user to debug.

I don't know if there is a generalized solution for catching this type of error.

A solution for this specific case would be to add `anchor_position` as a property and raise an error on it.

bradanlane, 2024-02-12:
I may be using `kwargs` incorrectly.

What I am saying is the following mistake does not generate an error:

```
from adafruit_display_text import label

tile = label.Label(font, text=msg, color=RED, scale=1, anchor_point=(0.0,0.0), anchor_position=(0, offset))
```

The result of this code is that `anchor_position` is ignored - aka has no effect - and results in the tile being at the top of the screen.

dhalbert, 2024-02-13:
We had a discussion about this in discord: unknown keyword args are being ignored. I suggested we really need to fix that.

tannewt, 2024-02-13:
I think this extra `**kwargs` is causing this: https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/blob/main/adafruit_display_text/__init__.py#L249 It gets anything that doesn't match and doesn't do anything with it.
I suspect removing it will fix this issue.

PR #212 (2024-11-15): Remove kwargs
@ladyada 

Resolves: #202 

Remove the `kwargs` and it's only usage. Incidentally it's only usage will effectively get done automatically now that we aren't using the wildcard.

Confirmed on a device that wrong arguments now raise an exception:
```
TypeError: unexpected keyword argument 'max_glyphs'
TypeError: unexpected keyword argument 'anchor_position'
```
closed 2024-11-15 by FoamyGuy