adafruit/Adafruit_CircuitPython_Display_Text #215: Not included in 20241116 build ?

YachtMango, 2024-11-16:
I've download the latest zip file ( adafruit-circuitpython-bundle-9.x-mpy-20241116) and can't see 'adafruit_display_text' lib.

FoamyGuy, 2024-11-16:
I just downloaded the 20241116 bundle from https://circuitpython.org/libraries and I do see this library in it.

Here is a screenshot showing it. 
[image]

I would say try downloading again and then double check. Ensure that the file explorer is not sorting by something unexpected causing them to appear in a different order.

YachtMango, 2024-11-16:
Thanks @FoamyGuy, User error as I was looking / searching for a file not a directory !! 

Issue closed

closed 2024-11-16 by YachtMango