adafruit/Adafruit_CircuitPython_Display_Text #221: TextBox not centering using ALIGN_CENTER and anchor_point/anchored_position

shubham0x13, 2025-05-27:
I'm having trouble getting `TextBox` to center text properly. Even though I'm using `align=TextBox.ALIGN_CENTER` along with `anchor_point` and `anchored_position`, the rendered text still appears off-center (shifted to the right).

Here's the code I'm using:

```py
import board
import displayio
import terminalio
from adafruit_display_text.text_box import TextBox
from busio import I2C
import adafruit_displayio_ssd1306

main_group = displayio.Group()

displayio.release_displays()
i2c = I2C(board.GP7, board.GP6)
display_bus = displayio.I2CDisplay(i2c, device_address=0x3C)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=64)

center_text = "Hello, World"
center_text_area = TextBox(
    terminalio.FONT,
    display.width,  # Maximum text width
    TextBox.DYNAMIC_HEIGHT,
    align=TextBox.ALIGN_CENTER,
    text=center_text,
    background_color=0x000000,
    color=0xFFFFFF,
    scale=1,
)

center_text_area.anchor_point = (0.5, 0.5)
center_text_area.anchored_position = (display.width // 2, display.height // 2)

main_group.append(center_text_area)

display.root_group = main_group

while True:
    pass
```

And here's the display output:
[image]

cinderblockgames, 2025-05-27:
Since you're making the width the whole display, you don't want the x coordinates to be centered, just the y.

Modified example:

```
center_text_area.anchor_point = (0, 0.5)
center_text_area.anchored_position = (0, display.height // 2)
```

FoamyGuy, 2025-05-27:
I tested this and confirmed there does seem to be a bug with anchor_point / anchored_position for TextBox. I am unsure of root cause yet, but can tell it's not behaving has intended.

PR #222 (2025-05-28): fix anchored positioning for TextBox
Resolves: #221 

`_reset_text()` was erroneously setting the `self._bounding_box` width to the value calculated dynamically instead of the user specified width value.  Which then caused the anchoring math to get thrown off. Height probably suffered from the same bug, but it would only matter if you were not using dynamic height.

This fixes it by setting the width and height to user specified values. I've also updated the TextBox example script to make use of anchor_point and anchored_position which is what I used for testing the fix.